tuya/TuyaOpen #7: 想要移植到esp32-s3，是否有移植说明手册

ruoxiaoye, 2024-04-30:


mhtcc520, 2024-05-22:
我们在5/30号有一个发布计划，会提供一份移植说明文档和工具。同时我们也有计划发布esp32系列芯片的board支持。

oksimple, 2024-08-22:
> We have a release scheduled for May 30th, during which we will provide a porting instruction document and tools. We also plan to release board support for the ESP32 series of chips ASAP.

Hi, is there any progress?

flyingcys, 2024-11-18:
Our Tuyaopen supports ESP32/ESP32C3 from v1.1.0, welcome to add other models via PR。
We provide relevant adaptation documentation：
[https://github.com/tuya/tuyaopen/blob/master/docs/en/contribute_guide.md](https://github.com/tuya/tuyaopen/blob/master/docs/en/contribute_guide.md)

chedecai, 2024-12-01:
[image]
从仓库中下载的，按照文档配置，分支切换到1.1.0，仅仅T2平台编译成功，编译T5/T3/esp32平台失败？

KyleDeng, 2024-12-25:
> ![2284afd8f92ac90627dfe44b66a9140](https://private-user-images.githubusercontent.com/108883812/391346649-c3729ea9-09b9-4411-8a42-0e41e6614bbb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUxMTUxNjIsIm5iZiI6MTczNTExNDg2MiwicGF0aCI6Ii8xMDg4ODM4MTIvMzkxMzQ2NjQ5LWMzNzI5ZWE5LTA5YjktNDQxMS04YTQyLTBlNDFlNjYxNGJiYi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyNVQwODIxMDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xOTFkMTU2MzJjNzBmZjk0OGJiMmQzYTFiMDQzNWVhYTlmMTMxMGZhZmNhMTc0MzBlNTBkMWExMTU2NTEyYjI3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.SikyagKrOASBqIu4wSEhsMqvooCW1QxENN6PZtpSFLU) 从仓库中下载的，按照文档配置，分支切换到1.1.0，仅仅T2平台编译成功，编译T5/T3/esp32平台失败？

如果方便，执行命令【tos check】，看一下工具版本是否正确。

flyingcys, 2024-12-25:
> ![2284afd8f92ac90627dfe44b66a9140](https://private-user-images.githubusercontent.com/108883812/391346649-c3729ea9-09b9-4411-8a42-0e41e6614bbb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJna [...]

可以尝试一下master上的代码，最新的代码我们加入了esp32镜像站点下载功能，可以更快的下载完成esp32对应的platform

closed 2025-06-23 by KyleDeng