tuya/TuyaOpen #12: Нe регистрируется продукт на ESP32-WROOM-32

AkagePiero, 2024-12-11:
Hello. I am using yours project to create the firmware. I'm trying to flash my module with the switch_demo project to test the module's interaction with the Tuya cloud and the Smart Life application. I have an ESP32-WROOM-32 Wi-Fi & Bluetooth module. Having successfully assembled the project using the required and supported platform and ESP32 chip, and having changed the configuration file, I uploaded the firmware using the ESP32 FLASH DOWNLOAD TOOL. Using the Terminal application, I made sure that the firmware works. In addition, my product is displayed in the Smart life application. But the connection ends with an error. I assume this is related to the errors shown in screenshot number 4. I conclude from the error -2 (invalid parameter) that the error is in the uuid or authkey. Can you help solve this problem, please?
Also, I saw a strange code that does not lead to anything good
In the switch_demo file (path: tuyaopen\apps\tuya_cloud\switch_demo\src\switch_demo.c) on line 217 there is a condition:
if (OPRT_OK != tuya_iot_license_read(&license))
Where is the tuya_iot_license_read function Defined in the tuyaopen\tools\porting\template\linux\tkl_flash.c file as:
int tuya_iot_license_read(tuya_iot_license_t *license)
{
    return OPRT_NOT_SUPPORTED;
}
It turns out that it will always write that the operation is not supported. There is no check for correct license
[image]
[image]
[image]
[image]
[20241211101257.txt](https://github.com/user-attachments/files/18092034/20241211101257.txt)

flyingcys, 2024-12-12:
Hi:

I looked at your serial port log and guessed that the reason you could not connect to Graffiti Cloud was because you did not use the correct authorization code for TuyaOpen.

You can obtain the correct TuyaOpen authorization code by following the [README.md](https://github.com/tuya/tuyaopen/blob/master/apps/tuya_cloud/README.md) instructions.

In addition, the UUID/AUTHKEY is an important identity for the device to connect to the graffiti cloud. It is not recommended to submit and display related content in the log to avoid related data leakage.

AkagePiero, 2024-12-12:
I received a license directly from the tuya development service by taking 2 free licenses to test the product

flyingcys, 2024-12-13:
We are now a free TuyaOpen authorization code, detailed information can have a look at this [https://github.com/tuya/tuyaopen/tree/master/apps/tuya_cloud#free-tuyaopen-authorization-code-giveaway](https://github.com/tuya/tuyaopen/tree/master/apps/tuya_cloud#free-tuyaopen-authorization-code-giveaway)

flyingcys, 2024-12-17:
Thank you so much for using our TuyaOpen project

We used the latest code from Tuyaopen github master, and tested on the ESP32-WROOME-32E，
 [image]

The ESP32 burning address is as follows
[image]

The test results can be connected to the network via Bluetooth and successfully connected to the tuya cloud

AkagePiero, 2024-12-17:
Thanks for the answer. I made sure that I flashed the device correctly. But the problem turned out to be related to the SmartLife application itself. The device works well in the TuyaSmart application, connects, I see the interface for interacting with the device, everything is fine. But through SmartLife it just gets stuck at the moment of data transfer via BLE

AkagePiero, 2024-12-18:
Thanks to all Tuya developers, the application has been updated and the firmware has started working on all applications.

closed 2024-12-18 by AkagePiero