tuya/TuyaOpen #14: How connect load this code in esp32

ManikandanVaradharaj, 2025-01-08:
I'm using esp32 dev board,Now I need to load this code and need to connect with Tuya App,for these How can i implement

ManikandanVaradharaj, 2025-01-08:
Check submodules.
TARGET=esp32
PROJ_NAME=sample_project_esp32
TARGET=esp32
CHIP=esp32
/home/smartbuild/tuyaopen/platform/esp32
Start board build_setup ...
IDF_PATH is empty
get_country_code ...
download esp-idf ...
Cloning into '/home/smartbuild/tuyaopen/platform/esp32/esp-idf'...
remote: Enumerating objects: 14886, done.
remote: Counting objects: 100% (14886/14886), done.
remote: Compressing objects: 100% (11621/11621), done.
error: RPC failed; curl 92 HTTP/2 stream 0 was not closed cleanly: CANCEL (err 8)
error: 942 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output
STSTUS###################################
-- Now is UNIX-like OS's.
-- [TOP] Using framework [base].
-- [TOP] PLATFORM_PATH: /home/smartbuild/tuyaopen/platform/esp32
-- [TOP] Using toolchain file [].
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:72 (project):
  The CMAKE_C_COMPILER:

    /home/smartbuild/tuyaopen/platform/esp32/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc

  is not a full path to an existing compiler tool.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:72 (project):
  The CMAKE_CXX_COMPILER:

    /home/smartbuild/tuyaopen/platform/esp32/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-g++

  is not a full path to an existing compiler tool.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "/home/smartbuild/tuyaopen/examples/get-started/sample_project/.build/sample_project_esp32/CMakeFiles/CMakeOutput.log".
See also "/home/smartbuild/tuyaopen/examples/get-started/sample_project/.build/sample_project_esp32/CMakeFiles/CMakeError.log".
make: *** No rule to make target 'example'.  Stop.

ManikandanVaradharaj, 2025-01-08:
96%] Built target tuyaos
Consolidate compiler generated dependencies of target tuyaapp
[ 97%] Built target tuyaapp
[ 98%] [EXAMPLE] Generate execute file [sample_project]
ERROR: /home/smartbuild/tuyaopen/platform/esp32/.espressif/python_env/idf5.2_py3.10_env/bin/python doesn't exist! Please run the install script or "idf_tools.py install-python-env" in order to create it
make[7]: *** [CMakeFiles/example_all.dir/build.make:76: build_example_exe] Error 1
make[6]: *** [CMakeFiles/Makefile2:507: CMakeFiles/example_all.dir/all] Error 2
make[5]: *** [CMakeFiles/Makefile2:514: CMakeFiles/example_all.dir/rule] Error 2
make[4]: *** [Makefile:173: example_all] Error 2
make[3]: *** [CMakeFiles/example.dir/build.make:70: CMakeFiles/example] Error 2
make[2]: *** [CMakeFiles/Makefile2:481: CMakeFiles/example.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:488: CMakeFiles/example.dir/rule] Error 2
make: *** [Makefile:160: example] Error 2

flyingcys, 2025-01-14:
The problem seems to be that not all esp-idf files on the platform have been downloaded. You can delete esp-idf files on the platform and retry the automatic download

Sindhuja-15, 2025-01-16:
@flyingcys which version should i need for these one.

ManikandanVaradharaj, 2025-01-16:
@ManikandanVaradharaj I installed the latest v5.4 of esp-idf,still it's getting the same error

ManikandanVaradharaj, 2025-01-16:
@flyingcys  Run build setup success ...
STSTUS###################################
-- Now is UNIX-like OS's.
-- [TOP] Using framework [base].
-- [KCONFIG] Copy pre-config to dot-config.
-- [KCONFIG] Use dot-config to generate cmake-config.
-- [KCONFIG] Use dot-config to generate header-file.
-- [TOP] PLATFORM_PATH: /home/smartbuild/tuyaopen/platform/esp32
-- [TOP] Using toolchain file [].
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/smartbuild/tuyaopen/platform/esp32/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/smartbuild/tuyaopen/platform/esp32/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using mbedtls from platform
-- [TOP] Using example [sample_project_esp32].
-- [TOP] PLATFORM_NEED_LIBS: tuyaapp tuyaos
-- [UT] Disable UT.
-- [TOP] If you want to build example
        [tos build].

-- [TOP] If you want to clean project
        [tos clean].

-- Configuring done
-- Generating done
-- Build files have been written to: /home/smartbuild/tuyaopen/examples/get-started/sample_project/.build/sample_project_esp32
[  0%] Building C object src/libhttp/CMakeFiles/libhttp.dir/src/http_client_wrapper.c.o
[  1%] Building C object src/libhttp/CMakeFiles/libhttp.dir/src/http_download.c.o
[  2%] Building C object src/libhttp/CMakeFiles/libhttp.dir/coreHTTP/source/core_http_client.c.o
[  2%] Building C object src/libhttp/CMakeFiles/libhttp.dir/coreHTTP/source/dependency/3rdparty/http_parser/http_parser.c.o
[  3%] Linking C static library ../../lib/liblibhttp.a
[  3%] Built target libhttp
[  3%] Building C object src/common/CMakeFiles/common.dir/backoffAlgorithm/source/backoff_algorithm.c.o
[  4%] Building C object src/common/CMakeFiles/common.dir/utilities/crc32i.c.o
[  4%] Building C object src/common/CMakeFiles/common.dir/utilities/crc_16.c.o
[  5%] Building C object src/common/CMakeFiles/common.dir/utilities/mix_method.c.o
[  5%] Building C object src/common/CMakeFiles/common.dir/utilities/uni_random.c.o
[  6%] Linking C static library ../../lib/libcommon.a
[  6%] Built target common
[  7%] Building C object src/tal_wifi/CMakeFiles/tal_wifi.dir/src/tal_wifi.c.o
/home/smartbuild/tuyaopen/src/tal_wifi/src/tal_wifi.c: In function 'tal_wifi_set_ip':
/home/smartbuild/tuyaopen/src/tal_wifi/src/tal_wifi.c:229:12: warning: implicit declaration of function 'tkl_wifi_set_ip'; did you mean 'tal_wifi_set_ip'? [-Wimplicit-function-declaration]
  229 |     return tkl_wifi_set_ip(wf, ip);
      |            ^~~~~~~~~~~~~~~
      |            tal_wifi_set_ip
[  7%] Linking C static library ../../lib/libtal_wifi.a
[  7%] Built target tal_wifi
[  8%] Building C object src/tal_kv/CMakeFiles/tal_kv.dir/src/tal_kv.c.o
In file included from /home/smartbuild/tuyaopen/src/tal_kv/littlefs/lfs_util.h:20,
                 from /home/smartbuild/tuyaopen/src/tal_kv/littlefs/lfs.h:11,
                 from /home/smartbuild/tuyaopen/src/tal_kv/include/tal_kv.h:26,
                 from /home/smartbuild/tuyaopen/src/tal_kv/src/tal_kv.c:27:
/home/smartbuild/tuyaopen/src/tal_kv/port/lfs_config.h:200:17: warning: 'lfs_crc' defined but not used [-Wunused-function]
  200 | static uint32_t lfs_crc(uint32_t crc, const void *buffer, size_t size)
      |                 ^~~~~~~
[ 10%] Building C object src/tal_kv/CMakeFiles/tal_kv.dir/src/kv_serialize.c.o
/home/smartbuild/tuyaopen/src/tal_kv/src/kv_serialize.c: In function 'kv_serialize':
/home/smartbuild/tuyaopen/src/tal_kv/src/kv_serialize.c:96:47: warning: format '%d' expects argument of type 'int', but argument 3 has type 'int32_t' {aka 'long int'} [-Wformat=]
   96 |             offset += sprintf(buf + offset, "%d", *((int32_t *)(db[i].val)));
      |                                              ~^   ~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                               |   |
      |                                               int int32_t {aka long int}
[... 627 lines cut ...]
[ 90%] Building C object src/tal_bluetooth/CMakeFiles/tal_bluetooth.dir/nimble/host/tuya_ble_mempool.c.o
[ 91%] Building C object src/tal_bluetooth/CMakeFiles/tal_bluetooth.dir/nimble/host/tuya_ble_os_adapter.c.o
[ 91%] Building C object src/tal_bluetooth/CMakeFiles/tal_bluetooth.dir/nimble/host/tuya_ble_peer.c.o
[ 92%] Building C object src/tal_bluetooth/CMakeFiles/tal_bluetooth.dir/nimble/host/tuya_hs_port.c.o
[ 93%] Building C object src/tal_bluetooth/CMakeFiles/tal_bluetooth.dir/nimble/modules/ble_eddystone.c.o
[ 93%] Building C object src/tal_bluetooth/CMakeFiles/tal_bluetooth.dir/nimble/modules/ble_ibeacon.c.o
[ 94%] Building C object src/tal_bluetooth/CMakeFiles/tal_bluetooth.dir/nimble/modules/tuya_ble_cli_svc.c.o
[ 94%] Building C object src/tal_bluetooth/CMakeFiles/tal_bluetooth.dir/nimble/tkl_bluetooth.c.o
[ 95%] Linking C static library ../../lib/libtal_bluetooth.a
[ 95%] Built target tal_bluetooth
[ 96%] Linking C static library lib/libtuyaos.a
[ 96%] Built target tuyaos
[... cut ...]
make[7]: *** [CMakeFiles/example_all.dir/build.make:76: build_example_exe] Error 1
make[6]: *** [CMakeFiles/Makefile2:507: CMakeFiles/example_all.dir/all] Error 2
make[5]: *** [CMakeFiles/Makefile2:514: CMakeFiles/example_all.dir/rule] Error 2
make[4]: *** [Makefile:173: example_all] Error 2
make[3]: *** [CMakeFiles/example.dir/build.make:70: CMakeFiles/example] Error 2
make[2]: *** [CMakeFiles/Makefile2:481: CMakeFiles/example.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:488: CMakeFiles/example.dir/rule] Error 2
make: *** [Makefile:160: example] Error 2

ManikandanVaradharaj, 2025-01-16:
@flyingcys  In examples/get-started/sample_project. project_build.ini 
I have set the platform and chip as esp32

[project:sample_project_esp32]
platform = esp32
chip = esp32

ManikandanVaradharaj, 2025-01-16:
@flyingcys /tuyaopen/examples/get-started/sample_project$ tos flash
--2025-01-16 14:27:10--  https://images.tuyacn.com/smart/embed/package/vscode/data/ide_serial/tyutool_cli.tar.gz
Resolving images.tuyacn.com (images.tuyacn.com)... 129.227.213.118, 101.33.16.239, 101.33.1.101, ...
Connecting to images.tuyacn.com (images.tuyacn.com)|129.227.213.118|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 9756584 (9.3M) [application/octet-stream]
Saving to: ‘tyutool_cli.tar.gz’

tyutool_cli.tar.gz                  100%[=================================================================>]   9.30M  4.91MB/s    in 1.9s    

2025-01-16 14:27:13 (4.91 MB/s) - ‘tyutool_cli.tar.gz’ saved [9756584/9756584]

tyutool params: 
[INFO]: tyut_logger init done.
[INFO]: Run Tuya Uart Tool.
Usage: tyutool_cli write [OPTIONS]
Try 'tyutool_cli write -h' for help.

Error: Invalid value for '-d' / '--device': 'esp32' is not one of 'BK7231N', 'T2', 'T3', 'T5', 'RTL8720CF', 'RTL8720CM', 'LN882H'.

ManikandanVaradharaj, 2025-01-17:
@flyingcys  In tuya tool.gui,I don't have the option of esp32 to flash.

ManikandanVaradharaj, 2025-01-20:
@flyingcys Can I use esp32 WROOM or esp32S3 module to connect with Tuya cloud

flyingcys, 2025-01-21:
> [@flyingcys](https://github.com/flyingcys) In tuya tool.gui,I don't have the option of esp32 to flash.

We are developing support for ESP32 series, should be supported by the next version conference, then choose the upgrade function on the software can be automatically upgraded.

flyingcys, 2025-01-21:
> [@flyingcys](https://github.com/flyingcys) Can I use esp32 WROOM or esp32S3 module to connect with Tuya cloud

@ManikandanVaradharaj 
Currently esp32/esp32c3 is supported, ESP32S3 has not been adapted and tested in detail for the time being, or can you help us implement ESP32S3?

flyingcys, 2025-01-21:
> [@flyingcys](https://github.com/flyingcys) which version should i need for these one.

The master branch already supports esp32/esp32c3
@Sindhuja-15

flyingcys, 2025-02-27:
> [@flyingcys](https://github.com/flyingcys) In tuya tool.gui,I don't have the option of esp32 to flash.

The tos flash command v1.8.x version already supports esp32 burning

closed 2025-06-23 by KyleDeng