tuya/TuyaOpen #17: ESP32-C3 Switch Demo WIFI connection failed

iBacklight, 2025-01-23:
Hi,

I'm trying to flash the switch demo to the esp32c3 dev board. I have successfully overwritten the uuid and the key of the `tuya_config.h` in the project (obtained from this giveaway [event](https://github.com/tuya/tuyaopen/blob/master/apps/tuya_cloud/README.md#free-tuyaopen-authorization-code-giveaway).) However, the output log always seems to showcase a "waiting for remote connection" status and I'm indeed connected to a 2.4G wifi point. In the Smart Life app, a connection window always said it failed in connection with tuya cloud. How can I solve this?

Here is a screenshot of smart life app:

[image]

Here is the full log file from esp32 end:

[image]

[image]

[image]

Thank you for your assistant!

flyingcys, 2025-01-23:
Hi
Can you check whether the app is the latest version, you can go to the App Store to upgrade to the latest version and try again

iBacklight, 2025-01-24:
> Thank you for your question. Can you check whether the app is the latest version, you can go to the App Store to upgrade to the latest version and try again

Hi, thank you for replying.

I'm pretty sure I'm using the latest version of the Smart Life app from the App Store. Are there any specific steps I missed, such as scanning a QR code? Currently, I open the app, and it just automatically reminds me if I would like to add the device or not.

iBacklight, 2025-02-04:
Hello, is there any update on this issue? I still cannot connect to the Smart Life app. Also there is no log information shown when I tried to connect my device to the cloud. The log will stuck on the pictures above.

Apart from that, I used esptool bootloader to flash the code to esp32c3. Will this be a part of the issue?

flyingcys, 2025-02-05:
I used the code on master to test that ESP32-C3 can connect to tuya cloud normally, whether you used the correct tuyaopen authorization code?

iBacklight, 2025-03-16:
I believe the issue lies in the product creation process on the Tuya Developer Platform. How can I correctly create a product that connects directly to the switch_demo project flashed on the ESP32-C3? The product I created isn't recognized by the Smart Life app. I followed the example in [Step 1 of the README](https://github.com/tuya/TuyaOpen/blob/master/apps/tuya_cloud/README.md) for the light demo, replicating its process closely, but it's still not working. Any insights?

[image]

danielmeza, 2025-04-19:
I buy 3 license in 10 days ago, still no way to download the licenses to place it in the ESP32, seems like Tuya doesn't want us to use third party devices but only Tuya propetary hardware.