tuya/TuyaOpen #19: CB3S and BK7231N programming

KonstaOrbanium, 2025-02-27:
Hello! I encountered a problem loading firmware into the CB3S module based on the bk7231n MK
The project compiles as it should, then, after my command, which the console advises me: .../tuyaopen/platform/bk7231x/tools/tyutool/cli write -d bk7231n -p [Port] -b [Baudrate] -f /home/vboxuser/tuyaopen/apps/tuya_cloud/switch_demo/.build/sample_project_bk7231x/bin/sample_project_bk7231x_QIO_1.0.0.bin - loading fails. The following outputs:

[INFO]: tyut_logger init done.
[INFO]: Run Tuya Uart Tool.
[INFO]: Use default start address: [0x00]
[INFO]: Waiting Reset ...
[WARNING]: Shake timeout!
[INFO]: Reboot done
[ERROR]: Flash write failed.

I specify the port available to me /dev/ttyUSB0 (this is a converter and it is in the list of USB devices), baudrate 921600 (I also tried to set different ones).
I assumed that one of the CSN pins (pin 12 or 19) is responsible for BOOT and before rebooting the controller I threw them to ground, but this did not help. I also tried to download the binary file through the BK7231GUIFlashTool program, but this also did not help
Please tell me how to flash BK7231N correctly?

I am attaching the file that I entered and what the console output to me

[cmd.txt](https://github.com/user-attachments/files/19002942/cmd.txt)

KyleDeng, 2025-02-27:
1. Please tell me the version of the cli tool you are using:./cli -v;
2. From the log, it can be seen that the tool sends a reset signal, but the module does not reply with valid content;
3. You can start the cli tool and wait for logs - [INFO]: Waiting Reset...
Then try to ground the reset pin a few times, more than 0.5s apart each time,

I hope I can help you

KyleDeng, 2025-02-27:
In addition, you can refer to the circuit used by the Doodle module, especially the reset circuit part.

https://developer.tuya.com/en/docs/iot-device-dev/T2-U-development-board?id=Kckeahvfhu7v0#title-13-Schematic%20diagram%20and%20PCB

closed 2025-03-03 by KonstaOrbanium