tuya/TuyaOpen #96: ESP32-S3 Thread Panic

gabrielpivatotramontina, 2025-04-29:
Hello, I've run the `tos set_example` and build the `get_started`  example, flashed it, and it ran just fine.
But when I was trying the `switch_demo` example, it builds and flashes ok, but it crashes while it's running.
 
Here is the output:
```
[01-01 00:00:00 ty D][tal_thread.c:236] Thread:sys_timer Exec Start. Set to Running Stat
[01-01 00:00:00 ty I][tal_thread.c:217] thread_create name:sys_timer,stackDepth:4096,totalstackDepth:4096,priority:5
[01-01 00:00:00 ty I][tal_thread.c:217] thread_create name:wq_system,stackDepth:6144,totalstackDepth:10240,priority:3
[01-01 00:00:00 ty D][tal_thread.c:236] Thread:wq_system Exec Start. Set to Running Stat
[01-01 00:00:00 ty D][tal_thread.c:236] Thread:wq_highpri Exec Start. Set to Running Stat
[01-01 00:00:00 ty I][tal_thread.c:217] thread_create name:wq_highpri,stackDepth:5120,totalstackDepth:15360,priority:4
assert failed: search_suitable_block tlsf_control_functions.h:359 (sl_map && "internal error - second level bitmap is null")
Backtrace: 0x40375d19:0x3fcaeee0 0x40381985:0x3fcaef00 0x403889d1:0x3fcaef20 0x40386cad:0x3fcaf040 0x40386865:0x3fcaf060 0x40376676:0x3fcaf080 0x40376695:0x3fcaf0b0 0x403762c5:0x3fcaf0d0 0x4202cf4f:0x3fcaf0f0 0x4202cf71:0x3fcaf110 0x42013601:0x3fcaf130 0x420184f2:0x3fcaf150 0x4200b6d1:0x3fcaf170 0x4200b735:0x3fcaf1c0 0x42008699:0x3fcaf1e0 0x42008753:0x3fcaf270 0x4201378e:0x3fcaf290
ELF file SHA256: f8c3a8dbf

```

And here is the decoded backtrace:
```
0x40375d19: panic_abort at /home/user/TuyaOpen/platform/ESP32/esp-idf/components/esp_system/panic.c:454
0x40381985: esp_system_abort at /home/user/TuyaOpen/platform/ESP32/esp-idf/components/esp_system/port/esp_system_chip.c:92
0x403889d1: __assert_func at /home/user/TuyaOpen/platform/ESP32/esp-idf/components/newlib/assert.c:80
0x40386cad: search_suitable_block at /home/user/TuyaOpen/platform/ESP32/esp-idf/components/heap/tlsf/tlsf_control_functions.h:359
0x40386cad: block_locate_free at /home/user/TuyaOpen/platform/ESP32/esp-idf/components/heap/tlsf/tlsf_control_functions.h:612
0x40386cad: tlsf_malloc at /home/user/TuyaOpen/platform/ESP32/esp-idf/components/heap/tlsf/tlsf.c:443
0x40386865: multi_heap_malloc_impl at /home/user/TuyaOpen/platform/ESP32/esp-idf/components/heap/multi_heap.c:209
0x40376676: aligned_or_unaligned_alloc at /home/user/TuyaOpen/platform/ESP32/esp-idf/components/heap/heap_caps_base.c:77
0x40376676: heap_caps_aligned_alloc_base at /home/user/TuyaOpen/platform/ESP32/esp-idf/components/heap/heap_caps_base.c:150
0x40376695: heap_caps_malloc_base at /home/user/TuyaOpen/platform/ESP32/esp-idf/components/heap/heap_caps_base.c:170
0x403762c5: heap_caps_malloc at /home/user/TuyaOpen/platform/ESP32/esp-idf/components/heap/heap_caps.c:84
0x4202cf4f: tkl_system_malloc at /home/user/TuyaOpen/platform/ESP32/tuya_open_sdk/tuyaos_adapter/src/system/tkl_memory.c:31
0x4202cf71: tkl_system_calloc at /home/user/TuyaOpen/platform/ESP32/tuya_open_sdk/tuyaos_adapter/src/system/tkl_memory.c:65
0x42013601: tal_calloc at /home/user/TuyaOpen/src/tal_system/src/tal_system.c:91
0x420184f2: tal_uart_init at /home/user/TuyaOpen/src/tal_driver/src/tal_uart.c:228
0x4200b6d1: tal_cli_init_with_uart at /home/user/TuyaOpen/src/tal_cli/src/tal_cli.c:774
0x4200b735: tal_cli_init at /home/user/TuyaOpen/src/tal_cli/src/tal_cli.c:813
0x42008699: user_main at /home/user/TuyaOpen/apps/tuya_cloud/switch_demo/src/tuya_main.c:219
0x42008753: tuya_app_thread at /home/user/TuyaOpen/apps/tuya_cloud/switch_demo/src/tuya_main.c:298
0x4201378e: __WrapRunFunc at /home/user/TuyaOpen/src/tal_system/src/tal_thread.c:238
```

Things I've tried so far: 
- Running `tos config_choice` again;
- Deleting `.build`;
- Running `tos clean`;
- Reinstalling the ESP32-S3 platform.

But no success so far. Could you please help me look into it?

gabrielpivatotramontina, 2025-05-07:
Thanks for the response.
Unfortunately I've ran in more problems

I did the following steps:
1. `git clone https://github.com/tuya/TuyaOpen.git`
2. `cd TuyaOpen/`
3. `git checkout release/v1.2.1`
4. `export PATH=$PATH:$PWD`
5. `tos set_example`
6. Choosed ESP32 examples option
7. `cd examples/get-started/sample_project/`
8. `tos build`

But got this:
```
In file included from /home/user/TuyaOpen/platform/ESP32/esp-idf/components/mbedtls/mbedtls/include/mbedtls/build_info.h:115,
                 from /home/user/TuyaOpen/platform/ESP32/esp-idf/components/mbedtls/mbedtls/include/mbedtls/rsa.h:20,
                 from /home/user/TuyaOpen/src/tal_security/include/tal_asymmetrical.h:36,
                 from /home/user/TuyaOpen/src/tal_security/include/tal_security.h:15,
                 from /home/user/TuyaOpen/src/tal_system/include/tal_api.h:59,
                 from /home/user/TuyaOpen/src/tal_bluetooth/src/tal_bluetooth.c:23:
/home/user/TuyaOpen/platform/ESP32/esp-idf/components/mbedtls/port/include/mbedtls/esp_config.h:28:10: fatal error: sdkconfig.h: No such file or directory
   28 | #include "sdkconfig.h"
      |          ^~~~~~~~~~~~~
compilation terminated.
[9/153] Building C object src/libtls/CMakeFiles/libtls.dir/src/cipher_wrapper.c.o
ninja: build stopped: subcommand failed.
```

I've tried some other workarounds, like getting the `sdkconfig.h` from another successful build, but still did not work.

flyingcys, 2025-05-08:
Thank you very much for using TuyaOpen. First, you conducted a test using apps/switch_demo. We will test and fix the issue you mentioned

gabrielpivatotramontina, 2025-05-08:
Right,but on both examples there is the same error, I reported now about the `sample_project` because it takes less time to compile.

flyingcys, 2025-05-16:
@gabrielpivatotramontina 
The latest master and release/v1.3.0 have both addressed esp32 IoT applications and AI applications

gabrielpivatotramontina, 2025-05-16:
Thanks for the effort you put in.

Unfortunately I've ran into some more errors (switch_demo):

```
I (516) main_task: Calling app_main()
[01-01 00:00:00 ty D][tal_thread.c:236] Thread:sys_timer Exec Start. Set to Running Stat
[01-01 00:00:00 ty I][tal_thread.c:217] thread_create name:sys_timer,stackDepth:4096,totalstackDepth:4096,priority:5
[01-01 00:00:00 ty I][tal_thread.c:217] thread_create name:wq_system,stackDepth:6144,totalstackDepth:10240,priority:3
[01-01 00:00:00 ty D][tal_thread.c:236] Thread:wq_system Exec Start. Set to Running Stat
[01-01 00:00:00 ty D][tal_thread.c:236] Thread:wq_highpri Exec Start. Set to Running Stat
         (586) uart: queue free spaces: 10
[01-01 00:00:00 ty I][tal_thread.c:217] thread_create name:cli,stackDepth:3072,totalstackDepth:22528,priority:2
[01-01 00:0
assert failed: search_suitable_block tlsf_control_functions.h:359 (sl_map && "internal error - second level bitmap is null")
Backtrace: 0x40375d21:0x3fcb3740 0x40381999:0x3fcb3760 0x403889e5:0x3fcb3780 0x40386cc1:0x3fcb38a0 0x40386879:0x3fcb38c0 0x4037667e:0x3fcb38e0 0x4037669d:0x3fcb3910 0x403762cd:0x3fcb3930 0x4202d083:0x3fcb3950 0x420136f0:0x3fcb3970 0x420189ca:0x3fcb39a0 0x42008b4c:0x3fcb3a50 0x42008bde:0x3fcb3a80 0x420087d6:0x3fcb3ab0 0x42008887:0x3fcb3b40 0x420138c2:0x3fcb3b60
ELF file SHA256: b1079b708
Rebooting...
``` 

Here is the decoded backtrace:
```
0x40375d21: panic_abort at /home/user/TuyaOpen/platform/ESP32/esp-idf/components/esp_system/panic.c:454
0x40381999: esp_system_abort at /home/user/TuyaOpen/platform/ESP32/esp-idf/components/esp_system/port/esp_system_chip.c:92
0x403889e5: __assert_func at /home/user/TuyaOpen/platform/ESP32/esp-idf/components/newlib/assert.c:80
0x40386cc1: search_suitable_block at /home/user/TuyaOpen/platform/ESP32/esp-idf/components/heap/tlsf/tlsf_control_functions.h:359
0x40386cc1: block_locate_free at /home/user/TuyaOpen/platform/ESP32/esp-idf/components/heap/tlsf/tlsf_control_functions.h:612
0x40386cc1: tlsf_malloc at /home/user/TuyaOpen/platform/ESP32/esp-idf/components/heap/tlsf/tlsf.c:443
0x40386879: multi_heap_malloc_impl at /home/user/TuyaOpen/platform/ESP32/esp-idf/components/heap/multi_heap.c:209
0x4037667e: aligned_or_unaligned_alloc at /home/user/TuyaOpen/platform/ESP32/esp-idf/components/heap/heap_caps_base.c:77
0x4037667e: heap_caps_aligned_alloc_base at /home/user/TuyaOpen/platform/ESP32/esp-idf/components/heap/heap_caps_base.c:150
0x4037669d: heap_caps_malloc_base at /home/user/TuyaOpen/platform/ESP32/esp-idf/components/heap/heap_caps_base.c:170
0x403762cd: heap_caps_malloc at /home/user/TuyaOpen/platform/ESP32/esp-idf/components/heap/heap_caps.c:84
0x4202d083: tkl_system_malloc at /home/user/TuyaOpen/platform/ESP32/tuya_open_sdk/tuyaos_adapter/src/system/tkl_memory.c:31
0x420136f0: tal_malloc at /home/user/TuyaOpen/src/tal_system/src/tal_system.c:53
0x420189ca: tal_kv_get at /home/user/TuyaOpen/src/tal_kv/src/tal_kv.c:276
0x42008b4c: reset_count_read at /home/user/TuyaOpen/apps/tuya_cloud/switch_demo/src/reset_netcfg.c:47
0x42008bde: reset_netconfig_start at /home/user/TuyaOpen/apps/tuya_cloud/switch_demo/src/reset_netcfg.c:102
0x420087d6: user_main at /home/user/TuyaOpen/apps/tuya_cloud/switch_demo/src/tuya_main.c:223
0x42008887: tuya_app_thread at /home/user/TuyaOpen/apps/tuya_cloud/switch_demo/src/tuya_main.c:298
0x420138c2: __WrapRunFunc at /home/user/TuyaOpen/src/tal_system/src/tal_thread.c:238
```

Could we work together a solution?

flyingcys, 2025-06-06:
Hi @gabrielpivatotramontina 
A new release version v1.3.1 has been released today. Could you please check again to see if the problem has been resolved?

gabrielpivatotramontina, 2025-06-06:
Hello,

gabrielpivatotramontina, 2025-06-06:
Hello,

Tried the new version, unfortunately it persists.
```
I (23) boot: ESP-IDF v5.4 2nd stage bootloader
I (23) boot: compile time Jun  6 2025 09:16:57
I (23) boot: Multicore bootloader
I (23) boot: chip revision: v0.2
I (26) boot: efuse block revision: v1.3
I (30) boot.esp32s3: Boot SPI Speed : 80MHz
I (33) boot.esp32s3: SPI Mode       : DIO
I (37) boot.esp32s3: SPI Flash Size : 8MB
I (41) boot: Enabling RNG early entropy source...
I (45) boot: Partition Table:
I (48) boot: ## Label            Usage          Type ST Offset   Length
I (54) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (61) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (67) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  3 ota_0            OTA app          00 10 00010000 00180000
I (80) boot:  4 ota_1            OTA app          00 11 00190000 00180000
I (87) boot:  5 tuya             Unknown data     01 aa 003b0000 00028000
I (93) boot:  6 factory_nvs      WiFi data        01 02 003e0000 00004000
I (100) boot: End of partition table
I (103) esp_image: segment 0: paddr=00010020 vaddr=3c0d0020 size=2bcf0h (179440) map
I (143) esp_image: segment 1: paddr=0003bd18 vaddr=3fc9da00 size=04300h ( 17152) load
I (146) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=ca954h (829780) map
I (294) esp_image: segment 3: paddr=0010a97c vaddr=3fca1d00 size=011c4h (  4548) load
I (295) esp_image: segment 4: paddr=0010bb48 vaddr=40374000 size=19940h (104768) load
I (320) esp_image: segment 5: paddr=00125490 vaddr=600fe100 size=0001ch (    28) load
I (330) boot: Loaded app from partition at offset 0x10000
I (331) boot: Disabling RNG early entropy source...
I (341) octal_psram: vendor id    : 0x0d (AP)
I (341) octal_psram: dev id       : 0x02 (generation 3)
I (341) octal_psram: density      : 0x03 (64 Mbit)
I (343) octal_psram: good-die     : 0x01 (Pass)
I (347) octal_psram: Latency      : 0x01 (Fixed)
I (352) octal_psram: VCC          : 0x01 (3V)
I (356) octal_psram: SRF          : 0x01 (Fast Refresh)
I (361) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (365) octal_psram: BurstLen     : 0x01 (32 Byte)
I (370) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (375) octal_psram: DriveStrength: 0x00 (1/1)
W (380) MSPI Timing: tuning fail, best point is fallen back to index 5
I (386) MSPI Timing: PSRAM timing tuning index: 5
I (390) esp_psram: Found 8MB PSRAM device
I (394) esp_psram: Speed: 80MHz
I (397) cpu_start: Multicore app
I (411) cpu_start: Pro cpu start user code
I (411) cpu_start: cpu freq: 160000000 Hz
I (412) app_init: Application information:
I (412) app_init: Project name:     tuya_open_sdk
I (416) app_init: App version:      v1.1.0-50-gd06d3ae-dirty
I (421) app_init: Compile time:     Jun  6 2025 09:16:38
I (426) app_init: ELF file SHA256:  a64f3d3e2...
I (431) app_init: ESP-IDF:          v5.4
I (434) efuse_init: Min chip rev:     v0.0
I (438) efuse_init: Max chip rev:     v0.99
I (442) efuse_init: Chip rev:         v0.2
I (446) heap_init: Initializing. RAM available for dynamic allocation:
I (452) heap_init: At 3FCA58B8 len 00043E58 (271 KiB): RAM
I (457) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (463) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (468) heap_init: At 600FE11C len 00001ECC (7 KiB): RTCRAM
I (473) esp_psram: Adding pool of 8178K of PSRAM memory to heap allocator
I (480) spi_flash: detected chip: gd
I (483) spi_flash: flash io: dio
I (486) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (492) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (499) coexist: coex firmware version: 49a8cdc
I (503) coexist: coexist rom version e7ae62f
[... 33 lines cut ...]
0x4037667a: aligned_or_unaligned_alloc at heap_caps_base.c:77
 (inlined by) heap_caps_aligned_alloc_base at heap_caps_base.c:150
0x40376699: heap_caps_malloc_base at heap_caps_base.c:170
0x403762c9: heap_caps_malloc at heap_caps.c:84
0x4202d14f: tkl_system_malloc at tkl_memory.c:33
0x4202d171: tkl_system_calloc at tkl_memory.c:68
0x42013801: tal_calloc at tal_system.c:91
0x420186f2: tal_uart_init at tal_uart.c:228
0x4200b8d5: tal_cli_init_with_uart at tal_cli.c:774
0x4200b939: tal_cli_init at tal_cli.c:813
0x420088a3: user_main at tuya_main.c:230
0x42008957: tuya_app_thread at tuya_main.c:309
0x4201398e: __WrapRunFunc at tal_thread.c:238
```

What envirorment are you using?

Mine is:
```
Distributor ID: Ubuntu
Description:    Ubuntu 24.04.2 LTS
Release:        24.04
Codename:       noble

```