tuya/TuyaOpen #145: t5适配ft6336触屏，I2c报错

kkcafe, 2025-05-23:
TOUCH_I2C_SCL 38
    TOUCH_I2C_SDA 39

    rt = tkl_io_pinmux_config(TOUCH_I2C_SCL, TUYA_IIC1_SCL);
    PR_DEBUG("pinmux %d to scl ret %d",TOUCH_I2C_SCL,rt);
    rt = tkl_io_pinmux_config(TOUCH_I2C_SDA, TUYA_IIC1_SDA);
    PR_DEBUG("pinmux %d to sda ret %d",TOUCH_I2C_SDA,rt);

    /*i2c init*/
    cfg.role = TUYA_IIC_MODE_MASTER;
    cfg.speed = TUYA_IIC_BUS_SPEED_100K;
    cfg.addr_width = TUYA_IIC_ADDRESS_7BIT;

    
    rt = tkl_i2c_init(TOUCH_I2C_PORT, &cfg);

  ....
       //#define FT6336_I2C_SLAVE_ADDR  (0x70 >> 1)
       //#define FT6336_CIPHER_REG        (0xA3)

       if (FT6336_i2c_port_read(FT6336_I2C_SLAVE_ADDR, FT6336_CIPHER_REG, &product_id, sizeof(product_id))) {
        PR_ERR("touch i2c read error");
        return -1;
    }

static int FT6336_i2c_port_read(uint16_t dev_addr, uint8_t register_addr, uint8_t *data_buf, uint16_t len)
{
    OPERATE_RET ret = OPRT_OK;
    uint8_t cmd_bytes = register_addr;
    //cmd_bytes[0] = (uint8_t)(register_addr >> 8);
    //cmd_bytes[1] = (uint8_t)(register_addr & 0x00FF);

    ret = tkl_i2c_master_send(TOUCH_I2C_PORT, dev_addr, &cmd_bytes, 1, FALSE);
    if (ret != OPRT_OK) {
        PR_ERR("send cmd fail %d",ret);
        return ret;
    }

    return tkl_i2c_master_receive(TOUCH_I2C_PORT, dev_addr, data_buf, len, FALSE);
}

最后报错：wait ack timeout，初始化失败。
请问一下，是哪里错了？

flyingcys, 2025-05-23:
可参考 [https://github.com/tuya/TuyaOpen/tree/master/src/peripherals/touch](https://github.com/tuya/TuyaOpen/tree/master/src/peripherals/touch) 定位问题

kkcafe, 2025-05-23:
参考了touch其他的代码，又参考了platform/T5AI/t5_os/components/bk_peripheral/src/tp/tp_ft6336.c
结果I2C报错，Pin和I2C port配置有问题吗？

shiliu-yang, 2025-05-23:
你这里报错的原因是模组没有收到 touch 的，可以通过逻辑分析仪抓下波形。看是 T5 模组I2C 驱动有问题，没能正确的发出波形；还是touch 的或者连线的问题

kkcafe, 2025-05-23:
感谢答复，确认是连线有问题。

closed 2025-05-26 by flyingcys