tuya/TuyaOpen #161: 构建失败：liblwip 中的 in_addr_t 类型定义冲突

TNTinMinecraft, 2025-06-01:
### 问题描述
在apps目录之外创建的项目无法通过 `tos build` 编译，错误发生在 `lwip_dhcpc.c` 原因是 `in_addr_t` 类型存在冲突定义。
### 错误详情
```
/home/tim/TuyaOpen/src/liblwip/lwip-2.1.2/src/include/lwip/inet.h:55:15: error: conflicting types for 'in_addr_t'
   55 | typedef u32_t in_addr_t;
      |               ^~~~~~~~~
In file included from /home/tim/TuyaOpen/platform/tools/gcc-arm-none-eabi-10.3-2021.10/arm-none-eabi/include/stdio.h:61,
                 from /home/tim/TuyaOpen/src/liblwip/lwip-2.1.2/src/include/lwip/arch.h:82,
                 from /home/tim/TuyaOpen/src/liblwip/lwip-2.1.2/src/include/lwip/debug.h:40,
                 from /home/tim/TuyaOpen/src/liblwip/lwip-2.1.2/src/include/lwip/opt.h:52,
                 from /home/tim/TuyaOpen/src/liblwip/lwip-2.1.2/src/include/lwip/mem.h:40,
                 from /home/tim/TuyaOpen/src/liblwip/port/lwip_dhcpc.c:14:
/home/tim/TuyaOpen/platform/tools/gcc-arm-none-eabi-10.3-2021.10/arm-none-eabi/include/sys/types.h:55:20: note: previous declaration of 'in_addr_t' was here
   55 | typedef __uint32_t in_addr_t; /* base type for internet address */
      |                    ^~~~~~~~~
```
### 系统环境
```
tim@tim-virtual-machine:~/hello$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.5 LTS
Release:        22.04
Codename:       jammy

tim@tim-virtual-machine:~/hello$ pwd
/home/tim/hello

tim@tim-virtual-machine:~/hello$ echo $PATH
/home/tim/.vscode-server/cli/servers/Stable-848b80aeb52026648a8ff9f7c45a9b0a80641e2e/server/bin/remote-cli:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/tim/TuyaOpen:

tim@tim-virtual-machine:~/hello$ uname -a
Linux tim-virtual-machine 6.8.0-59-generic #61~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Apr 15 17:03:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
```

### 完整编译输出
```
tim@tim-virtual-machine:~/hello$ tos build
/home/tim/hello
Check [bash](5.1.16) > [4.0.0]: OK.
Check [grep](3.7) > [3.0.0]: OK.
Check [sed](4.8) > [4.0.0]: OK.
Check [python3](3.10.12) > [3.6.0]: OK.
Check [git](2.34.1) > [2.0.0]: OK.
Check [ninja](1.10.1) > [1.6.0]: OK.
Check [cmake](3.22.1) > [3.16.0]: OK.
Check submodules.
enable git mirror...
disable git mirror...
/home/tim/TuyaOpen/platform/T5AI
Start board build_setup ...
/home/tim/TuyaOpen/platform/T5AI
Start board prepare ...
Toolchain /home/tim/TuyaOpen/platform/T5AI/../tools/gcc-arm-none-eabi-10.3-2021.10 check Successful
Run board prepare success ...
Run build setup success ...
-- ###################################
-- Now is UNIX-like OS's.
-- [TOP] Using framework [base].
-- OS: Linux

-- [TOP] Using toolchain file [].
-- [TOP] Using example [hello].
-- [TOP] PLATFORM_NEED_LIBS: tuyaapp tuyaos
-- EXAMPLE_LIB:/home/tim/hello
-- [UT] Disable UT.
-- [TOP] If you want to build example
        [tos build].

[... 9 lines cut ...]
   40 | static void tuya_app_thread(void *arg)
      |                             ~~~~~~^~~
[50/308] Building C object src/tal_bluetooth/CMakeFiles/tal_bluetooth.dir/nimble/host/ble_svc_gap.c.o
/home/tim/TuyaOpen/src/tal_bluetooth/nimble/host/ble_svc_gap.c:49:38: warning: 'ble_svc_gap_defs' defined but not used [-Wunused-const-variable=]
   49 | static const struct ble_gatt_svc_def ble_svc_gap_defs[] = {
      |                                      ^~~~~~~~~~~~~~~~
[63/308] Building C object src/liblwip/CMakeFiles/liblwip.dir/port/lwip_dhcpc.c.o
FAILED: src/liblwip/CMakeFiles/liblwip.dir/port/lwip_dhcpc.c.o 
/home/tim/TuyaOpen/platform/T5AI/../tools/gcc-arm-none-eabi-10.3-2021.10/bin/arm-none-eabi-gcc  -I/home/tim/TuyaOpen/src/liblwip/lwip-2.1.2/src/include -I/home/tim/TuyaOpen/src/liblwip/lwip-2.1.2/src/include/lwip -I/home/tim/TuyaOpen/src/liblwip/lwip-2.1.2/src/include/lwip/apps -I/home/tim/TuyaOpen/src/liblwip/lwip-2.1.2/src/include/compat -I/home/tim/hello/.build/include -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/adc -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/asr -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/asr/bat_asr -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/asr/libtutuClear -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/asr/wanson_asr -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/bluetooth -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/display -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/flash -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/gpio -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/hci -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/i2c -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/init -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/init/include -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/init/src -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/ipc -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/media -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/network -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/pinmux -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/pwm -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/rtc -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/rtc/doc -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/rtc/include -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/security -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/spi -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/spi/doc -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/spi/include -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/system -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/timer -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/uart -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/utilities -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/utilities/include -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/utilities/src -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/vad -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/vad/lib -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/wakeup -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/wakeup/doc -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/wakeup/include -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/watchdog -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/wifi -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/include/wired -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/src -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/src/driver -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/src/driver/tkl_display -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/src/misc -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/src/system -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/src/test -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/src/test/test_lvgl -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/src/test/test_lvgl/stress -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/src/test/test_lvgl/widgets -I/home/tim/TuyaOpen/platform/T5AI/tuyaos/tuyaos_adapter/src/test/test_lvgl/widgets/assets -I/home/tim/TuyaOpen/platform/T5AI/t5_os/bk_idk/include -I/home/tim/TuyaOpen/platform/T5AI/t5_os/build/bk7258_cp1/config -I/home/tim/TuyaOpen/src/tal_driver/include -I/home/tim/TuyaOpen/src/libhttp/include -I/home/tim/TuyaOpen/src/libhttp/coreHTTP -I/home/tim/TuyaOpen/src/libhttp/coreHTTP/source/include -I/home/tim/TuyaOpen/src/libhttp/coreHTTP/source/dependency/3rdparty/http_parser -I/home/tim/TuyaOpen/src/tal_bluetooth/include -I/home/tim/TuyaOpen/src/tal_bluetooth/nimble/include -I/home/tim/TuyaOpen/src/tuya_ai_basic/include -I/home/tim/TuyaOpen/src/peripherals/audio_codecs/tdl_audio/include -I/home/tim/TuyaOpen/src/peripherals/button/tdd_button_driver/include -I/home/tim/TuyaOpen/src/peripherals/button/tdl_button_manage/include -I/home/tim/TuyaOpen/src/tal_system/include -I/home/tim/TuyaOpen/src/tal_network/include -I/home/tim/TuyaOpen/src/tal_security/include [... cut ...]
                 from /home/tim/TuyaOpen/src/liblwip/lwip-2.1.2/src/include/lwip/mem.h:40,
                 from /home/tim/TuyaOpen/src/liblwip/port/lwip_dhcpc.c:14:
/home/tim/TuyaOpen/platform/tools/gcc-arm-none-eabi-10.3-2021.10/arm-none-eabi/include/sys/types.h:55:20: note: previous declaration of 'in_addr_t' was here
   55 | typedef __uint32_t in_addr_t; /* base type for internet address */
      |                    ^~~~~~~~~
[76/308] Building C object src/tuya_ai_basic/CMakeFiles/tuya_ai_basic.dir/src/tuya_ai_protocol.c.o
ninja: build stopped: subcommand failed.
```

TNTinMinecraft, 2025-06-03:
> 请问你新建的项目，选择的是什么platform

T5AI

shiliu-yang, 2025-06-04:
> > 你可以尝试下先使用 `tos config_choice` 选择一下配置，然后调用 `tos fullclean` ，再进行编译 `tos build` 试试。 目前部分 examples 默认配置有点问题，问题现象和你这里的一样，正在修改。
> 
> 使用 `tos config_choice` 选择配置后，会覆盖之前在 `tos menuconfig` 中选择的配置。 只使用 `tos menuconfig` 配置后，项目中并没有正确配置platform，同时 `.build/cache/using.config` 和 `.using_config` 中的 `liblwip` 部分也没有正确配置。

对于一个同一个芯片平台，他的 lwip、tls.. 这些配置基本上都是一样的，所以可以先使用 `tos config_choice` 先将默认配置修改为对应平台的默认配置。然后如果还有一些需要特殊修改，在  `tos config_choice` 之后再调用  `tos menuconfig` 对那部分进行修改，修改完成后再调用  `tos savedef` 将修改后的配置进行保存。

`.using_config` 这些文件的配置都是依赖于 menuconfig 进行生成的，tuyaopen 支持不同厂商的不同的芯片 ，在创建工程的时候不知道开发者所使用的芯片具体是哪一个，不同芯片的配置也是差别很大的。这里为了减少开发者的配置所以预先设置了一个默认配置，需要调用 `tos config_choice` 进行一个默认配置的选择，这个命令会覆盖之前的配置。

所以这里创建一个新的工程，正确操作应该是：

1. `tos config_choice` 选择使用的芯片平台的默认配置
2. `tos menuconfig` ，非必选，按照实际情况对默认配置进行修改
3. `tos savedef`，保存当前配置