tuya/TuyaOpen #204: macos上 TuyaOpen/apps/tuya.ai /your_chat_bot/ 编译 lwip 出错

HonestQiao, 2025-06-21:
## 环境和配置
```
$ sw_vers
ProductName:	macOS
ProductVersion:	12.7.6
BuildVersion:	21H1320

$ tos version
[INFO]: Running tos.py ...
[INFO]: v1.3.0-26-g94f4346

$ pwd
/Users/HonestQiao/Projects/tuya/TuyaOpen/apps/tuya.ai/your_chat_bot

$ tos config choice
[INFO]: Running tos.py ...
[INFO]: [do subprocess]: cd /Users/HonestQiao/Projects/tuya/TuyaOpen/apps/tuya.ai/your_chat_bot/.build && ninja clean_all
[0/2] [TOP] Platform [T5AI] clean:
Cleaning successful for T5AI.
[2/2] cd /Users/HonestQiao/Projects/tuya/TuyaOpen/apps/tuya.ai/your_chat_bot/.build && /Users/HonestQiao/Projects/tuya/TuyaOpen/.venv/bin/ninja clean
[1/1] Cleaning all built files...
Cleaning... 421 files.
[INFO]: Clean success.
[INFO]: Fullclean success.
--------------------
1. TUYA_T5AI_MINI_LCD_1.3.config
2. WAVESHARE_ESP32S3_TOUCH_AMOLED_1_8.config
3. DNESP32S3_BOX.config
4. DNESP32S3.config
5. TUYA_T5AI_BOARD_EYES.config
6. TUYA_T5AI_BOARD_LCD_3.5.config
7. ESP32S3_BREAD_COMPACT_WIFI.config
8. TUYA_T5AI_EVB.config
9. XINGZHI_Cube_0_96OLED_WIFI.config
10. T5AI_MOJI_1.28.config
--------------------
Input "q" to exit.
Choice config file: 6
```

## 编译
```
$ tos  build
[INFO]: Running tos.py ...
[INFO]: Initialing using.config ...
[INFO]: Platform [T5AI] is exists.
[INFO]: Platform [T5AI] downloaded successfully.
[INFO]: Preparing platform [T5AI] ...
[INFO]: [do subprocess]: cd /Users/HonestQiao/Projects/tuya/TuyaOpen/platform/T5AI && python platform_prepare.py T5AI
platform_root: /Users/HonestQiao/Projects/tuya/TuyaOpen/platform/tools
country code: Hong Kong
get package from [Hong Kong darwin_x86_64]
[Toolchain folder is exists]: /Users/HonestQiao/Projects/tuya/TuyaOpen/platform/tools/arm-gnu-toolchain-13.3.rel1-darwin-x86_64-arm-none-eabi
[INFO]: Platform [T5AI] prepared successfully.
[INFO]: Build setup for [your_chat_bot] success.
[INFO]: [do subprocess]: cd /Users/HonestQiao/Projects/tuya/TuyaOpen/apps/tuya.ai/your_chat_bot/.build && cmake -G Ninja /Users/HonestQiao/Projects/tuya/TuyaOpen -DTOS_PROJECT_NAME=your_chat_bot -DTOS_PROJECT_ROOT=/Users/HonestQiao/Projects/tuya/TuyaOpen/apps/tuya.ai/your_chat_bot -DTOS_PROJECT_PLATFORM=T5AI -DTOS_FRAMEWORK=base -DTOS_PROJECT_CHIP=T5AI -DTOS_PROJECT_BOARD=TUYA_T5AI_BOARD
-- #################### TuyaOpen CMakeLists.txt Start ####################
-- Now is Apple systens.
-- [TOP] System processor: x86_64
-- [TOP] Using framework [base].
-- [KCONFIG] Use dot-config to generate cmake-config.
-- [KCONFIG] Use dot-config to generate header-file.
-- [TOP] Using toolchain file [].
-- The C compiler identification is GNU 13.3.1
-- The CXX compiler identification is GNU 13.3.1
-- Checking whether C compiler has -isysroot
-- Checking whether C compiler has -isysroot - yes
-- Checking whether C compiler supports OSX deployment target flag
-- Checking whether C compiler supports OSX deployment target flag - no
-- Detecting C compiler ABI info
[... 186 lines cut ...]
/Users/HonestQiao/Projects/tuya/TuyaOpen/src/liblwip/lwip-2.1.2/src/api/sockets.c:166:85: note: in definition of macro 'LWIP_SOCKOPT_CHECK_OPTLEN'
  166 | #define LWIP_SOCKOPT_CHECK_OPTLEN(sock, optlen, opttype) do { if ((optlen) < sizeof(opttype)) { done_socket(sock); return EINVAL; }}while(0)
      |                                                                                     ^~~~~~~
/Users/HonestQiao/Projects/tuya/TuyaOpen/src/liblwip/lwip-2.1.2/src/api/sockets.c:3532:11: note: in expansion of macro 'LWIP_SOCKOPT_CHECK_OPTLEN_CONN'
 3532 |           LWIP_SOCKOPT_CHECK_OPTLEN_CONN(sock, optlen, LWIP_SO_SNDRCVTIMEO_OPTTYPE);
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/HonestQiao/Projects/tuya/TuyaOpen/src/liblwip/lwip-2.1.2/src/api/sockets.c:3532:56: note: in expansion of macro 'LWIP_SO_SNDRCVTIMEO_OPTTYPE'
 3532 |           LWIP_SOCKOPT_CHECK_OPTLEN_CONN(sock, optlen, LWIP_SO_SNDRCVTIMEO_OPTTYPE);
      |                                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/HonestQiao/Projects/tuya/TuyaOpen/src/liblwip/lwip-2.1.2/src/api/sockets.c:203:80: error: invalid use of undefined type 'const struct timeval'
  203 | #define LWIP_SO_SNDRCVTIMEO_GET_MS(optval) ((((const struct timeval *)(optval))->tv_sec * 1000) + (((const struct timeval *)(optval))->tv_usec / 1000))
      |                                                                                ^~
/Users/HonestQiao/Projects/tuya/TuyaOpen/src/liblwip/lwip-2.1.2/src/api/sockets.c:3533:21: note: in expansion of macro 'LWIP_SO_SNDRCVTIMEO_GET_MS'
 3533 |           ms_long = LWIP_SO_SNDRCVTIMEO_GET_MS(optval);
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/HonestQiao/Projects/tuya/TuyaOpen/src/liblwip/lwip-2.1.2/src/api/sockets.c:203:134: error: invalid use of undefined type 'const struct timeval'
  203 | #define LWIP_SO_SNDRCVTIMEO_GET_MS(optval) ((((const struct timeval *)(optval))->tv_sec * 1000) + (((const struct timeval *)(optval))->tv_usec / 1000))
      |                                                                                                                                      ^~
/Users/HonestQiao/Projects/tuya/TuyaOpen/src/liblwip/lwip-2.1.2/src/api/sockets.c:3533:21: note: in expansion of macro 'LWIP_SO_SNDRCVTIMEO_GET_MS'
 3533 |           ms_long = LWIP_SO_SNDRCVTIMEO_GET_MS(optval);
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~
[21/315] Building C object src/liblwip/CMakeFiles/liblwip.dir/lwip-2.1.2/src/core/ipv4/dhcp.c.o
ninja: build stopped: subcommand failed.
[ERROR]: Build error.
```

KyleDeng, 2025-06-23:
由于mac系统并没有arm-gcc-10.3的支持，所以在mac中使用了arm-gcc-14.2，
gcc版本差异，在T5芯片使用时，需要做如下修改：
1. config choice选择T5对应的开发板
2. config menu打开配置界面
3. 搜索配置项 IN_ADDR_T_DEFINED（键盘输入"/addr"并回车），取消 IN_ADDR_T_DEFINED
4. 搜索配置项 LWIP_TIMEVAL_PRIVATE（键盘输入"/timeval"并回车），改 LWIP_TIMEVAL_PRIVATE 为1
5. 保存配置

HonestQiao, 2025-06-23:
> 由于mac系统并没有arm-gcc-10.3的支持，所以在mac中使用了arm-gcc-14.2， gcc版本差异，在T5芯片使用时，需要做如下修改：
> 
> 1. config choice选择T5对应的开发板
> 2. config menu打开配置界面
> 3. 搜索配置项 IN_ADDR_T_DEFINED（键盘输入"/addr"并回车），取消 IN_ADDR_T_DEFINED
> 4. 搜索配置项 LWIP_TIMEVAL_PRIVATE（键盘输入"/timeval"并回车），改 LWIP_TIMEVAL_PRIVATE 为1
> 5. 保存配置

测试通过，多谢啦！

flyingcys, 2025-06-27:
https://github.com/tuya/TuyaOpen/pull/216
已经修复改问题，可以正常编译。