tuya/TuyaOpen #205: macos 上 tos flash 运行出错

HonestQiao, 2025-06-21:
环境同 #204 
your_chat_bot_QIO_1.0.1.bin 是我在Ubuntu环境编译好的。

## 烧录
```
tos flash
[INFO]: Running tos.py ...
[INFO]: Downloading tyutool_cli ...
[INFO]: Flash command: /Users/HonestQiao/Projects/tuya/TuyaOpen/tools/tyutool/tyutool_cli write -d T5AI -f /Users/HonestQiao/Projects/tuya/TuyaOpen/apps/tuya.ai/your_chat_bot/.build/bin/your_chat_bot_QIO_1.0.1.bin
[INFO]: [do subprocess]: /Users/HonestQiao/Projects/tuya/TuyaOpen/tools/tyutool/tyutool_cli write -d T5AI -f /Users/HonestQiao/Projects/tuya/TuyaOpen/apps/tuya.ai/your_chat_bot/.build/bin/your_chat_bot_QIO_1.0.1.bin
[10617] Error loading Python lib '/var/folders/4n/psszlnqn73x1vwzl44cq258h0000gn/T/_MEIoQpaX1/Python': dlopen: dlopen(/var/folders/4n/psszlnqn73x1vwzl44cq258h0000gn/T/_MEIoQpaX1/Python, 0x000A): Symbol not found: (_mkfifoat)
  Referenced from: '/private/var/folders/4n/psszlnqn73x1vwzl44cq258h0000gn/T/_MEIoQpaX1/Python'
  Expected in: '/usr/lib/libSystem.B.dylib'
[ERROR]: Flash failed.
```

adwuard, 2025-06-22:
Python 怎么会从系统里面找?
有 . ./export.sh 使能一下Python虚拟环境吗?

HonestQiao, 2025-06-22:
> Python 怎么会从系统里面找? 有 . ./export.sh 使能一下Python虚拟环境吗?

另外，这个错误，是因为tos flash实际调用了TuyaOpen/tools/tyutool/tyutool_cli，这个工具，可能应该是个独立的工具。

KyleDeng, 2025-06-23:
这个问题是由于涂鸦在macOS 系统上使用 PyInstaller 打包的 Python 应用程序时与你的mac环境并不完全相同，
打包环境和运行环境之间的兼容性问题导致的。

KyleDeng, 2025-06-23:
可以尝试使用windows或者linux中的烧录工具，后续涂鸦会考虑使用python源码方式完成烧录步骤，
解决跨平台产物不兼容的类似问题，
感谢你的反馈，我们会加快修复

HonestQiao, 2025-06-23:
> 这个问题是由于涂鸦在macOS 系统上使用 PyInstaller 打包的 Python 应用程序时与你的mac环境并不完全相同， 打包环境和运行环境之间的兼容性问题导致的。

那我要在什么样的macOS系统环境中，才能正常运行呢？

@KyleDeng

adwuard, 2025-06-23:
@KyleDeng 用户的 Issue 问题还没正式反馈解决，我们这边不主动关闭哈。

HonestQiao, 2025-06-23:
我这边，直接使用tyutool源码，能够成功运行了。

解决方案如下：
```
$ cd TuyaOpen/tools/tyutool

$ git clone https://github.com/tuya/tyutool.git

$ cat > tyutool_cli <<EOF
#!/bin/bash

# 使用 python 执行当前目录下的 tyutool_cli.py ，并确保在任何目录都能调用

# 设置为当前目录下的tyutool，用绝对路径
TYUTOOL_CLI_PATH=\$(cd "\$(dirname "\$0")" && pwd)/tyutool

# 执行Python脚本并传递所有参数
python "\${TYUTOOL_CLI_PATH}/tyutool_cli.py" "\$@"
EOF

$ pip install rich tqdm
```

然后，就可以正常运行：
```
cd TuyaOpen/apps/tuya.ai/your_chat_bot

# 烧录
tos flash

# 监听
tos monitor
```

HonestQiao, 2025-06-23:
> 我这边，直接使用tyutool源码，能够成功运行了。
>
> 解决方案如下：
>
> ```
> $ cd TuyaOpen/tools/tyutool
>
> $ git clone https://github.com/tuya/tyutool.git
>
> $ cat > tyutool_cli <<EOF
> #!/bin/bash
>
> # 使用 python 执行当前目录下的 tyutool_cli.py [...]

测试验证，T5 AI开发板和ESP32S3_BREAD_COMPACT_WIFI，都可以正常烧录和监听

adwuard, 2025-07-01:
这边看问题已经验证成功一周了，先关闭Issue。有其他问题可以再Re-open。谢谢 @HonestQiao 反馈和贡献