tuya/TuyaOpen #206: dev分支在Windos下编译成功，无法正常配网

molun, 2025-06-21:
[image]

HonestQiao, 2025-06-22:
类似的情况：ESP32S3_BREAD_COMPACT_WIFI，编译烧录后，也无法配网添加设备。

## 环境：
系统环境： #204 
硬件环境：ESP32S3_BREAD_COMPACT_WIFI
```
$ esptool flash_id
esptool.py v4.9.0
Found 1 serial ports
Serial port /dev/cu.wchusbserial1420
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: cc:ba:97:21:a7:14
Uploading stub...
Running stub...
Stub running...
Manufacturer: 68
Device: 4018
Detected flash size: 16MB
Flash type set in eFuse: quad (4 data lines)
Flash voltage set by eFuse to 3.3V
Hard resetting via RTS pin...
```

具体测试步骤如下：

## 配置

```
$ tos config choice
[INFO]: Running tos.py ...
[INFO]: Fullclean success.
--------------------
1. TUYA_T5AI_MINI_LCD_1.3.config
2. WAVESHARE_ESP32S3_TOUCH_AMOLED_1_8.config
3. DNESP32S3_BOX.config
4. DNESP32S3.config
5. TUYA_T5AI_BOARD_EYES.config
6. TUYA_T5AI_BOARD_LCD_3.5.config
7. ESP32S3_BREAD_COMPACT_WIFI.config
8. TUYA_T5AI_EVB.config
9. XINGZHI_Cube_0_96OLED_WIFI.config
10. T5AI_MOJI_1.28.config
--------------------
Input "q" to exit.
Choice config file: 7

```

## 编译
```
$ tos build
...
Activating ESP-IDF 5.4
* Checking python version ... 3.12.4
* Checking python dependencies ... OK
* Deactivating the current ESP-IDF environment (if any) ... OK
* Establishing a new ESP-IDF environment ... OK
* Identifying shell ... bash
* Detecting outdated tools in system ... OK - no outdated tools found
* Shell completion ... Autocompletion code generated
WARNING: Failed to load shell autocompletion for bash version: 3.2.57(1)-release!

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build
esptool.py v4.9.0
Wrote 0xf170e2 bytes to file /Users/HonestQiao/Projects/tuya/TuyaOpen/platform/ESP32/tuya_open_sdk/build/your_chat_bot_QIO_1.0.1.bin, ready to flash to offset 0x0
[INFO]: ******************************
[INFO]: /Users/HonestQiao/Projects/tuya/TuyaOpen/apps/tuya.ai/your_chat_bot/.build/bin/your_chat_bot_QIO_1.0.1.bin
[INFO]: ******************************
[INFO]: ******* Build Success ********
[INFO]: ******************************
```

## 烧录
```
$ esptool write_flash -z 0x00 /Users/HonestQiao/Projects/tuya/TuyaOpen/apps/tuya.ai/your_chat_bot/.build/bin/your_chat_bot_QIO_1.0.1.bin
esptool.py v4.9.0
Found 1 serial ports
Serial port /dev/cu.wchusbserial1420
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: cc:ba:97:21:a7:14
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00f17fff...
Compressed 15823076 bytes to 1776074...
Wrote 15823076 bytes (1776074 compressed) at 0x00000000 in 187.7 seconds (effective 674.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

## 涂鸦app发现和配网：

[image]

[image]

[image]

## 串口监听配网过程
```
$ picocom -b 115200 /dev/cu.wchusbserial1420
...
[01-01 00:00:46 ty I][tkl_bluetooth.c:146] BLE_GAP_EVENT_CONNECT(0x00), handle = 0x01, Role(1)

[01-01 00:00:46 ty D][ble_hs_hci_evt.c:152] evt code null:20
[01-01 00:00:46 ty N][ble_mgr.c:909] Ble Connected
[01-01 00:00:46 ty I][ble_hs.c:393] ble_hs_timer_exp:
[01-01 00:00:46 ty D][ble_att_svr.c:1936] debug rc = 0
[01-01 00:00:46 ty I][tkl_bluetooth.c:181] BLE_GAP_EVENT_CONN_UPDATE_REQ
[01-01 00:00:47 ty I][tkl_bluetooth.c:216] BLE_GAP_EVENT_SUBSCRIBE
[01-01 00:00:47 ty N][tkl_bluetooth.c:177] BLE_GAP_EVENT_CONN_UPDATE,0x6,0x0,0x1f4
[01-01 00:00:47 ty I][tkl_bluetooth.c:181] BLE_GAP_EVENT_CONN_UPDATE_REQ
[01-01 00:00:47 ty N][tkl_bluetooth.c:177] BLE_GAP_EVENT_CONN_UPDATE,0x28,0x0,0x1f4
[01-01 00:00:47 ty I][tkl_bluetooth.c:190] mtu update event; conn_handle=0x0001 mtu=246, channel id = 4

[01-01 00:00:47 ty D][ble_mgr.c:226] ble recv sub_pkg desc:3, no:1, pack_len:33, total_len:33
[01-01 00:00:47 ty E][ble_mgr.c:297] ble packet len err:64092
[01-01 00:00:47 ty E][ble_mgr.c:938] tuya_ble_data_proc fail. -2
[01-01 00:00:48 ty I][tkl_bluetooth.c:216] BLE_GAP_EVENT_SUBSCRIBE
[01-01 00:00:48 ty N][tkl_bluetooth.c:158] BLE_GAP_EVENT_DISCONNECT(0x213)

[01-01 00:00:48 ty I][ble_gap.c:2248] GAP init: stop adv
[01-01 00:00:48 ty I][ble_hs.c:433] ble_hs_timer reset:0
[01-01 00:00:48 ty E][ble_mgr.c:360] ret:-28683
[01-01 00:00:48 ty N][ble_gap.c:2509] Start Adv
[01-01 00:00:48 ty N][ble_mgr.c:365] ble adv updated 0
[01-01 00:00:48 ty N][ble_mgr.c:923] Ble Disonnected
[01-01 00:00:48 ty I][ble_hs.c:393] ble_hs_timer_exp:
[01-01 00:00:51 ty I][app_system_info.c:64] Free heap size:67307
[01-01 00:01:01 ty I][app_system_info.c:64] Free heap size:67307
[01-01 00:01:11 ty I][app_system_info.c:64] Free heap size:67307
```

## 小智AI运行情况
可正常运行从源码编译的小智AI，并完成配网操作
```
$ grep BOARD sdkconfig
# CONFIG_BOARD_TYPE_CUSTOM is not set
CONFIG_BOARD_TYPE_BREAD_COMPACT_WIFI=y
```

HonestQiao, 2025-06-22:
添加对应的TUYA_PRODUCT_KEY、TUYA_OPENSDK_UUID、TUYA_OPENSDK_AUTHKEY后，重新编译烧录，问题解决。

adwuard, 2025-06-22:
好的 谢谢即时反馈。👏 
@shiliu-yang config 文件补充添加上，添加完就吧这个Issue close 掉吧。