tuya/TuyaOpen #229: macOS 上 TuyaOpen 构建失败，`arm-none-eabi-gcc` 提示无法识别 `-arch` 标志

AnShuda, 2025-07-02:
## 问题描述
在 macOS 上为 T5AI 平台构建 `your_chat_bot` 示例（使用 `TUYA_T5AI_BOARD_LCD_3.5.config` 配置）时，构建过程在 `ninja example` 阶段失败。`arm-none-eabi-gcc` 编译器报错，提示无法识别 `-arch x86_64` 标志，而该标志不适用于 ARM 交叉编译。

## 环境信息
- **操作系统**：macOS Sequoia 15.5
- **电脑型号**：MacBook Air 13-inch, M3, 2024
- **芯片**：Apple M3
- **内存**：8 GB
- **Python**：3.13.5（Homebrew 安装）
- **Git**：2.50.0（Homebrew 安装）
- **Make**：4.4.1（Homebrew 安装）
- **CMake**：4.0.2（pip 安装，虚拟环境）
- **Ninja**：1.11.1.4（pip 安装，虚拟环境）
- **工具链**：ARM GNU Toolchain 13.3.rel1（平台准备阶段下载）
- **仓库**：TuyaOpen（`https://github.com/tuya/TuyaOpen.git`）
- **平台**：T5AI
- **板卡配置**：`TUYA_T5AI_BOARD_LCD_3.5.config`
- **项目**：`your_chat_bot`
- **TuyaOpen 版本**：v1.4.0（通过 `tos.py version` 确认）

## 复现步骤
1. 安装依赖：
   ```bash
   brew install python3 git make
   ```
2. 克隆仓库：
   ```bash
   git clone https://github.com/tuya/TuyaOpen.git
   cd TuyaOpen
   ```
3. 设置环境：
   ```bash
   . ./export.sh
   ```
4. 验证环境：
   ```bash
   tos.py version  # 输出 v1.4.0
   tos.py check    # 确认依赖和下载子模块
   ```
5. 进入 `your_chat_bot` 示例并配置：
   ```bash
   cd apps/tuya.ai/your_chat_bot
   tos.py config choice  # 选择选项 6：TUYA_T5AI_BOARD_LCD_3.5.config
   ```
6. 构建项目：
   ```bash
   tos.py build
   ```
7. 构建在 `ninja example` 阶段失败。

## 预期行为
构建成功，生成 `your_chat_bot` 应用的二进制文件，适用于 T5AI 平台。

## 实际行为
构建失败，多个源文件编译时 `arm-none-eabi-gcc` 报错：
```
arm-none-eabi-gcc: error: unrecognized command-line option '-arch'; did you mean '-march='?
```
受影响的文件包括：
- `reset_netcfg.c`
- `app_display.c`
- `tuya_main.c`
- `font_awesome_*.c`
- `apps/tuya.ai/your_chat_bot/src/` 下的其他文件

## 日志
```
[1/705] Building C object CMakeFiles/tuyaapp.dir/apps/tuya.ai/your_chat_bot/src/reset_netcfg.c.o
FAILED: CMakeFiles/tuyaapp.dir/apps/tuya.ai/your_chat_bot/src/reset_netcfg.c.o 
.../arm-none-eabi-gcc ... -arch x86_64 ... -c /Users/hughchu/TuyaOpen/apps/tuya.ai/your_chat_bot/src/reset_netcfg.c
arm-none-eabi-gcc: error: unrecognized command-line option '-arch'; did you mean '-march='?
...
ninja: build stopped: subcommand failed.
[ERROR]: Build error.
```

## 额外背景
- `-arch x86_64` 是 macOS 原生构建的标志，不适用于 ARM 交叉编译。
- 问题可能由 CMake 配置（`CMakeLists.txt` 或相关脚本）或工具链设置（`platform_prepare.py`、`tos.py`）错误引入 macOS 特定标志。
- ARM GNU 工具链（13.3.rel1）在 `platform_prepare.py` 阶段下载和解压正常。
- CMake 配置阶段无错误，问题仅出现在 Ninja 构建阶段。

## 建议修复
- 检查 CMake 配置（`CMakeLists.txt` 或相关脚本），确保不将 `-arch x86_64` 传递给 `arm-none-eabi-gcc`。
- 验证工具链文件或构建脚本（`platform_prepare.py`、`tos.py`），排除 macOS 特定标志。
- 考虑在交叉编译时添加目标架构检查，防止传递无效标志。

## 尝试的解决方法
- 未修改源码或构建脚本，因不确定正确修复方法。
- 清理后重新运行 `tos.py config choice` 和 `tos.py build`，仍报相同错误。

请指导如何解决此问题，或确认是否为 TuyaOpen 构建系统的 bug。如需更多日志或信息，请告知。感谢涂鸦团队的支持！

KyleDeng, 2025-07-04:
@AnShuda 
1.请使用tos.py version命令
2.执行在platform/T5AI中执行git log -1
请将这两个信息提供一下，方便我们定位问题

KyleDeng, 2025-07-04:
目前我有疑问的是，在M3架构下，应该会识别出，PROCEASSOR=“arm64”，
但是在你的环境中并没有
可否在你的环境中执行如下命令：
python
import platform
platform.machine()

然后将日志发我看下

AnShuda, 2025-07-04:
> 目前我有疑问的是，在M3架构下，应该会识别出，PROCEASSOR=“arm64”， 但是在你的环境中并没有 可否在你的环境中执行如下命令： python import platform platform.machine()
> 
> 然后将日志发我看下

[image]

我这边测试了一下，在终端中执行 platform.machine() 返回的是 'x86_64'，说明当前 Python 环境是走的 Rosetta 模拟模式（Intel 架构），不是原生的 arm64。

可能正是这个原因导致 PROCESSOR 识别不到为 arm64。

我会重新开一个 原生 arm64 的终端（关闭 Rosetta 模式），再切回原生 Python 再试一遍构建。

AnShuda, 2025-07-09:
已确认目前构建成功，问题原因及解决方案总结如下：

之前构建失败是由于系统中安装了 x86_64 架构的 Python（Rosetta 模拟模式下运行），导致 `platform.machine()` 返回 `x86_64`，而不是预期的 `arm64`，从而影响了构建脚本对平台架构的识别与执行。

为解决该问题，我采取了以下措施：
	1.	**删除旧的 x86_64 架构 Python**，确保环境中只存在 Apple Silicon 原生的 arm64 架构 Python。
	2.	**通过命令 `python3 -c "import platform; print(platform.machine())"` 验证**，返回 `arm64`，说明已使用原生架构。
	3.	由于 Tuya 的构建脚本默认调用 `python` 命令（而不是 `python3`），因此我在 `/usr/local/bin/` 下写了一个 shim，转发 `python` 到 `python3`，如下：
```bash
#!/bin/bash
exec /usr/bin/python3 "$@"
```

并通过 `chmod +x /usr/local/bin/python` 赋予执行权限。

现在 `python --version` 能正确调用原生 Python 3，`platform.machine()` 返回 `arm64`，Tuya 项目也可以成功编译和烧录。因此，该 issue 可以关闭，感谢支持！

[image]

[image]

[image]

closed 2025-07-09 by KyleDeng