tuya/TuyaOpen #256: lvgl_demo代码修改

chiyikouxian, 2025-07-12:
#if LV_USE_DEMO_WIDGETS

#if LV_MEM_CUSTOM == 0 && LV_MEM_SIZE < (38ul * 1024ul)
    #error Insufficient memory for lv_demo_widgets. Please set LV_MEM_SIZE to at least 38KB (38ul * 1024ul).  48KB is recommended.
#endif

/*********************
 *      DEFINES
 *********************/

/**********************
 *      TYPEDEFS
 **********************/
typedef enum {
    DISP_SMALL,
    DISP_MEDIUM,
    DISP_LARGE,
} disp_size_t;

/**********************
 *  STATIC PROTOTYPES
 **********************/
static void profile_create(lv_obj_t * parent);
static void analytics_create(lv_obj_t * parent);
static void shop_create(lv_obj_t * parent);
static void color_changer_create(lv_obj_t * parent);

static lv_obj_t * create_meter_box(lv_obj_t * parent, const char * title, const char * text1, const char * text2,
                                   const char * text3);
static lv_obj_t * create_shop_item(lv_obj_t * parent, const void * img_src, const char * name, const char * category,
                                   const char * price);

static void color_changer_event_cb(lv_event_t * e);
static void color_event_cb(lv_event_t * e);
static void ta_event_cb(lv_event_t * e);
static void birthday_event_cb(lv_event_t * e);
static void calendar_event_cb(lv_event_t * e);
static void slider_event_cb(lv_event_t * e);
static void chart_event_cb(lv_event_t * e);
static void shop_chart_event_cb(lv_event_t * e);
static void meter1_indic1_anim_cb(void * var, int32_t v);
static void meter1_indic2_anim_cb(void * var, int32_t v);
static void meter1_indic3_anim_cb(void * var, int32_t v);
static void meter2_timer_cb(lv_timer_t * timer);
static void meter3_anim_cb(void * var, int32_t v);

/**********************
 *  STATIC VARIABLES
 **********************/
static disp_size_t disp_size;

static lv_obj_t * tv;
static lv_obj_t * calendar;
static lv_style_t style_text_muted;
static lv_style_t style_title;
static lv_style_t style_icon;
static lv_style_t style_bullet;

static lv_obj_t * meter1;
static lv_obj_t * meter2;
static lv_obj_t * meter3;

static lv_obj_t * chart1;
static lv_obj_t * chart2;
static lv_obj_t * chart3;

static lv_chart_series_t * ser1;
static lv_chart_series_t * ser2;
static lv_chart_series_t * ser3;
static lv_chart_series_t * ser4;

static const lv_font_t * font_large;
static const lv_font_t * font_normal;

static uint32_t session_desktop = 1000;
static uint32_t session_tablet = 1000;
static uint32_t session_mobile = 1000;

这是tuyaopen框架下的platform >T5Al > t5_os > projects > Ivgl > widgets > main > C Iv_demo_widgets.c > profile_create(lv_obj_t *)里面的代码，我想要进入到第一行代码以下的代码，问题出在代码段的第一行，我把所有的LV_USE_DEMO_WIDGETS定义全改成1了，但是还是无法进入第一行之后的代码，我想知道是什么问题

chiyikouxian, 2025-07-14:
那platform >T5Al > t5_os > projects > Ivgl这个路径下的工程文件可以当作例程来进行编译和使用吗

chiyikouxian, 2025-07-14:
我现在可以正常编译examples里的lvgl_demo，并且烧录到T5AI这块板子上，我想要设计我需要的UI界面，我应该在哪个文件目录下进行修改

yingtao450, 2025-07-14:
如果是编写你自己的ui 界面，你可以在lvgl_demo中或者自己新建一个应用工程，直接调用lvgl的接口实现你的界面。如果你想修改lvgl 的官方demo, lvgl的源码或者配置头文件，你可以在src/liblvgl/文件夹下进行修改。

yingtao450, 2025-07-15:
1. 你的vscode 如果跳转到platform >T5Al > t5_os > projects > Ivgl > widgets > main > C Iv_demo_widgets.c 这个路径下，是你的vscode 跳转错地方了。
2.我前面也回复的了，真正的liblvgl的路径是在src/liblvgl 下，即你要寻找的任何跟lvgl库相关的修改都在这个路径下。
3.你想要修改lv_demo_whight 这个函数，这个函数的路径在src/liblvgl/lvgl/demos/widgets/lv_demo_widgets.c中。
4.至于你的UI设计如果是依赖lvgl的，可以自行在网上搜索lvgl的使用教程，或者阅读lvgl官网手册。

closed 2025-07-15 by chiyikouxian