tuya/TuyaOpen #259: ESP32 in Tuya OS has ethernet support

henrique-Backes, 2025-07-14:
I need a Wi-Fi, BLE, and Ethernet (ETH) connection with the ESP32, and I'm having trouble using Ethernet. I'm using Windows 11 with an OLIMEX ESP32 Gateway board, which comes with a LAN8720 and an RJ45 connector. Is there a ready-made solution available, or should I adapt a new one?

flyingcys, 2025-07-15:
Currently, we do not have a wired Ethernet driver available for ESP32. The Ubuntu platform currently uses wired Ethernet. It can be bound through scanning. You can refer to the demo apps/tuya_cloud/switch_demo for more information.

henrique-Backes, 2025-07-15:
Which board do you suggest I use as a reference to create a new ESP32 board with wired connection support? I saw that the T5-AI supports wired connections, or should I use Ubuntu as a reference instead? The port for wired connection support already exists in ESP-IDF, so I think I can add this functionality.